The-OpenROAD-Project/OpenSTA #135: Doc: in the "get_pins" command reference, idiom -filter "direction == output" does not work

tcal-x, 2023-02-17:
I found that the spaces were causing the problem;  `-filter "direction == output"` causes
```
Error: sta_by_stage.tcl line 85, unsupported pin -filter expression.
```
but `-filter "direction==output"` works as expected.

jjcherry56, 2023-02-26:
I cannot reproduce this. You need to include a complete testcase that demonstrates the issue.
In my tests it works correctly.
report_object_full_names [get_ports -filter "direction == input" *]
clk1
clk2
in
report_object_full_names [get_ports -filter "direction==input" *]
clk1
clk2
in
report_object_full_names [get_pins -filter "direction == input" *]
b1/clk
b1/in
b2/clk
b2/in
% report_object_full_names [get_pins -filter "direction==input" *]
report_object_full_names [get_pins -filter "direction==input" *]
b1/clk
b1/in
b2/clk
b2/in

tcal-x, 2023-02-27:
Thanks @jjcherry56!   Let me check which version we have -- maybe we need an update.

tcal-x, 2023-02-27:
I should have at least given the full `get_pints` command:   I use both `-of_objects` and `-filter`, which may be why your testcase did not encounter it:
```
set qp [get_pins -of_objects $inst -filter “direction == output”]
```
Output (using version 2.2.1) includes:
```
Warning: tp-sta.tcl line 36, patterns argument not supported with -of_objects.
Error: tp-sta.tcl line 36, unsupported pin -filter expression.
```

Full test case attached: [osta-135.tar.gz](https://github.com/The-OpenROAD-Project/OpenSTA/files/10842896/osta-135.tar.gz)

tcal-x, 2023-02-27:
The last line here is where I found this idiom in the docs (OpenSTA.pdf):

[image]

jjcherry56, 2023-02-27:
It is some very strange tcl behavior that fails to group the quoted filter expression into one token in the proc args.
It works with {}'s or without the spaces. I'll update the doc but there isn't a code change that will
overcome the real issue.

jjcherry56, 2023-02-27:
I don't see the motivation for the script; it seems to be reverse engineering report_checks (which does not print any paths because the clock is not defined so there are no constrained paths). You can also do
report_checks -unconstrained -format summary
to get something close to what the script is reporting if you don't define the clock.

jjcherry56, 2023-02-28:
The problem is non-printing characters in your get_pins expression that causes token breaks by the tcl parser. The spaces in the filter expression are not the problem. You might want to look for a new text editor.

﻿﻿from od -t a on your command file

﻿﻿﻿﻿﻿﻿﻿﻿0002040   sp  sp  sp  sp  sp   s   e   t  sp   q   p  sp   [   g   e   t
0002060    _   p   i   n   s  sp   -   o   f   _   o   b   j   e   c   t
0002100    s  sp   $   i   n   s   t  sp   -   f   i   l   t   e   r  sp
0002120    ?  80  9c   d   i   r   e   c   t   i   o   n  sp   =   =  sp
0002140    o   u   t   p   u   t   ?  80  9d   ]  nl  sp  sp  sp  sp  sp

closed 2023-02-28 by jjcherry56

tcal-x, 2023-02-28:
@jjcherry56 thanks for finding that and sorry for the hassle.

It must have come from a copy and paste, from GitHub in a chrome browser to Vim running in xterm -- the double quotes had been changed into fancy directional ones  -- these are the (bad) directional ones although it's not so apparent when you view them here: “direction==output”

zerzerzerz, 2023-03-02:
Can I dump the output of `report_object_full_names` to a file like `report_checks > report.log`? Thanks!

zerzerzerz, 2023-03-02:
```tcl
with_output_to_variable input_ports {report_object_full_names [get_ports * -filter "direction==input"]}
echo $input_ports > input_ports 
```

tcal-x, 2023-03-02:
@zerzerzerz it's probably best to open a new issue with your question.

However I will point you to the `redirect_file_begin` and `redirect_file_end` commands, so that you could do
```
redirect_file_begin "report.log"
report_object_full_names [get_ports -filter "direction == input" *]
redirect_file_end
```
(not tested)