The-OpenROAD-Project/OpenSTA #152: compilation with SSTA=1 fails

aviadb, 2023-03-26:
Hi,
No documentation is provided for the SSTA variable.
However, digging in the source tree I see that SSTA is sued for POCV calcs. 

`[tcl/StaTcl.i]`
0204 StaTcl.i:4072             POCV support requires compilation with SSTA=1.

I set the SSTA in the cmake cmd like this: -DSSTA=1
and I get the following compilation error:

/home/aviadb/apps/apps_src/OpenSTA/dcalc/LumpedCapDelayCalc.cc:168:59:   required from here
/usr/include/c++/12/cmath:632:5: error: no type named ‘__type’ in ‘struct __gnu_cxx::__enable_if<false, bool>’
/usr/include/c++/12/cmath:624:3: note: candidate: ‘constexpr bool std::isnan(long double)’
  624 |   isnan(long double __x)
      |   ^~~~~
/usr/include/c++/12/cmath:624:21: note:   no known conversion for argument 1 from ‘const sta::Slew’ {aka ‘const sta::Delay’} to ‘long double’
  624 |   isnan(long double __x)
      |         ~~~~~~~~~~~~^~~
/usr/include/c++/12/cmath:619:3: note: candidate: ‘constexpr bool std::isnan(double)’
  619 |   isnan(double __x)
      |   ^~~~~
/usr/include/c++/12/cmath:619:16: note:   no known conversion for argument 1 from ‘const sta::Slew’ {aka ‘const sta::Delay’} to ‘double’
  619 |   isnan(double __x)
      |         ~~~~~~~^~~
/usr/include/c++/12/cmath:611:3: note: candidate: ‘constexpr bool std::isnan(float)’
  611 |   isnan(float __x)
      |   ^~~~~
/usr/include/c++/12/cmath:611:15: note:   no known conversion for argument 1 from ‘const sta::Slew’ {aka ‘const sta::Delay’} to ‘float’
  611 |   isnan(float __x)
      |         ~~~~~~^~~
make[2]: *** [CMakeFiles/OpenSTA.dir/build.make:299: CMakeFiles/OpenSTA.dir/dcalc/LumpedCapDelayCalc.cc.o] Error 1`

jjcherry56, 2023-03-26:
It isn't documented because it an experimental feature for one customer and I am not really interested in supporting it for general use at this time. I fixed the compile errors but don't expect 
20455b5 SSTA compile errors on https://github.com/jjcherry56/OpenSTA that OR will eventually update to.

closed 2023-03-26 by jjcherry56

aviadb, 2023-03-30:
Thank you @jjcherry56 , I managed to compile after the fix.
I'm interested in testing that feature...
Will you accept bug tickets and pull requests for it?

jjcherry56, 2023-03-30:
Who are you?

aviadb, 2023-03-31:
Who am I?
Is that really matters?

Is OpenSTA open for pull-requests and bug tickets?

jjcherry56, 2023-03-31:
I am the sole author of opensta and have worked on the code base for 23 years.
If you expect me to spend my time fixing bugs and reviewing pull requests for you I think
they least you can do is tell me who I am working for free to help.