The-OpenROAD-Project/OpenSTA #161: VCD Based power analysis is failing when the design has memories

sakundu, 2023-04-22:
I am attempting to run VCD-based power analysis for Ariane on NG45 using OpenROAD/OpenSTA, and I encountered the following error.

`[ERROR STA-0803] ./inputs/ariane.vcd line 553579, Unknown variable type integer.`

I have tried running two different commercial tools and received the same output from both. Therefore, I believe my inputs are correct. I have uploaded all the necessary inputs required to reproduce the aforementioned error in my google drive ([Link](https://drive.google.com/file/d/1UZoDvuoA2A__sUcAeJPvn1moSXSim6c3/view?usp=share_link)).

```
tar -xvf ariane_ng45.tar.gz
openroad ./inputs/power_vcd_or.tcl
``` 
I successfully ran VCD-based power analysis for designs without memories using OpenROAD and didn't encounter any errors.

If you could fix this issue that would be great. Please let me know if you need any more information.

Thanks,
Sayak

tspyrou, 2023-04-22:
@abk-openroad and @sakundu I can reproduce this problem. Thanks for packaging it up.
The line in question is 
$var integer 32 _c= j [31:0] $end

jjcherry56, 2023-04-24:
fixed in https://github.com/jjcherry56/OpenSTA.git e5c95ac issue161 vcd var type integer

commit a139d6b6 (2025-01-02): get_* do not require liberty resolves #161