The-OpenROAD-Project/OpenSTA #165: About driving resistance and the Arnoldi model order reduction

gzz2000, 2023-05-25:
I know this should be better posted as a discussion. I have a technical question regarding the driving resistance implementation in the reduced order model. In OpenSTA, the original R and C matrices are first gone through the Krylov subspace computation. After that, the driving resistance is computed, and added to the top-leftmost element in the reduced order model matrix (the 5x5 matrix H) after multiplied by the (total capacitance)[https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/dcalc/ArnoldiDelayCalc.cc#L609]. This mysteriously works very well and my experiment shows that the only changed element is the top-leftmost element, and it is indeed linearly dependent on the applied driving resistance. I wonder if there is any mathematical reference of this property. Thanks for your help and your elegant open-source Arnoldi implementation!

jjcherry56, 2023-05-28:
The only documentation of the arnoldi delay calculator implementation is in dcalc/Arnoldi.txt

closed 2023-05-28 by jjcherry56

gzz2000, 2023-06-02:
Hi James, thanks for pointing me to the documentation. 
Do you know how to contact the original author William Scott? It seems some technical details are not present there and I want to ask if he could kindly give me some hints. Thank you.

jjcherry56, 2023-06-02:
I do not know how to get in touch with him. If you find anything additional please let me know.