The-OpenROAD-Project/OpenSTA #166: "No paths found" for corners over 8 corners

donn, 2023-06-01:
```
OpenSTA 2.4.0 71a707440d Copyright (c) 2021, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>
```

When defining 8 timing corners, all of them work. But when defining 9, one breaks, and when defining 10, two break, both reporting "No paths found."

[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenSTA/files/11624324/repro.tar.gz)

jjcherry56, 2023-06-01:
It has a bit field that overflows with more than 8 corners, so treat that has an unchecked limitation for the time being.

jjcherry56, 2023-06-07:
fixed in commit https://github.com/parallaxsw/OpenSTA.git
aeb6027 James Cherry 2023-06-01 increase max corner count to 128

commit 80ecdf26 (2025-01-21): ReportTcl const resolves #166
closed 2023-06-07 by jjcherry56