The-OpenROAD-Project/OpenSTA #168: Question: SDF operating conditions

kareefardi, 2023-06-15:
Do I need to always call `set_operating_conditions` to vary the conditions in SDF written using `write_sdf`. If yes then without `set_operating_condtions` what are the conditions used during SDF write (especially with multiple corners defined).

jjcherry56, 2023-06-15:
write_sdf does not really "use" the operating conditions. Delay calculation does, and write_sdf merely writes the delay calculation results to a file.

set_operating_conditions is unfortunately rather overloaded so the answer depends on what set
of arguments you pass it, what libraries exist and how the corners are defined. I am not really willing to enumerate what happens in all of the possible cases. I think the only situation that it makes any sense to use set_operating_conditions is if you only have one library and are trying to model having separate libraries for multiple corners. If you have separate libraries for each corner, you can derate them separately with -library argument but it doesn't make a lot of sense because the pvt is baked into the liberty values.

jjcherry56, 2023-06-16:
With no -corner arg write_add min/max delays are across all corners.

kareefardi, 2023-06-18:
What I meant are these values written by sdf:
```
 (VOLTAGE 1.600::1.600)
 (PROCESS "1.000::1.000")
 (TEMPERATURE 100.000::100.000)
```
Without `set_operating_conditions`, these values don't change when specifying `-corner`. They only changed when `set_operating_conditions` was explicitly called. For example, this is roughly what I am doing:
```
define_corners <corner1> <corner2>
read_lib -corner <corner1> file1
read_lib -corner <corner2> file2
read_verilog ...
link_design <design>
read_sdc <sdc_file>
write_sdf -corner <corner1> <sdf_file1>
write_sdf -corner <corner2> <sdf_file2>
```
`<sdf_file1>` and `<sdf_file2>` have same values for `VOLTAGE` `PROCESS` and `TEMPERATURE`. I can create a test case if needed.

kareefardi, 2023-07-04:
@jjcherry56 Do you need a test case or more information ?

jjcherry56, 2023-09-18:
commit 2054d01 in https://github.com/parallaxsw/OpenSTA.git writes the corner library default operating conditions pvt

closed 2023-09-18 by jjcherry56