The-OpenROAD-Project/OpenSTA #172: Not getting the actual names of inputs,outputs,registers in sta.

Vinayakamk, 2023-06-26:
INTENTION: to set path groups, so we need list of inputs ,registers and outputs from sta tool after reading sythesised verilog.

now,after reading liberty,verilog and linking the design ,all_inputs,all_outputs,all_registers cmd gives list renamed intances,
why it isnt showing actual names of inputs,registers,outputs as specified in verilog?

**{kindy see the preview  for image and attached yosys generated  verilog file in txt as extention. }**

[image]

[mk_combo_mul.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/11869541/mk_combo_mul.txt)
 sythesised verilog

rovinski, 2023-06-26:
Those commands return pointers to the objects. If you want the names, then use `get_name`. I don't know if there's a command that works on a list, but a for loop will work:
```
openroad> all_registers
_90dec8996d550000_p_Instance _00dfc8996d550000_p_Instance _70dfc8996d550000_p_Instance _e0dfc8996d550000_p_Instance

openroad> foreach reg [all_registers] {puts [get_name $reg]}
_672_
_673_
_674_
_675_
_676_
```

jjcherry56, 2023-06-26:
set_path_group arguments accept the result of all_registers or all_inputs like any other sta command. That is the whole point of SDC. The printed result is irrelevant. get_full_name is more appropriate if the netlist has hierarchy.

Vinayakamk, 2023-06-27:
thanx  for replying u both.
my problem is what could be the issue for this  design (mk_combo_mul ) that i couldnot able to set path group apart **from input to output path group.**

but i could get **all the path groups for the picorv32** which is available code in **openlane design section.**

.sdc file is attached for setting up path group..
[mk_combo_mul_sdc.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/11885783/mk_combo_mul_sdc.txt)

kindly check where i went wrong
should the liberty file should be choosen which supports all this or something else?

jjcherry56, 2023-06-27:
I can't decode your message but it is pretty obvious that the command file does not use sdc commands correctly. It looks like you cut/pasted the output of all_inputs/all_registers
commands into the command file.
The correct way to call the command is as a tcl function that returns the value:

group_path -name i2o -from [all_inputs] -to [all_outputs]

or something similar.

Vinayakamk, 2023-07-01:
one progress i want to share is, i could do flop2output pathgrup **outside docker** (without 'make mount' inside openlane) and only input2flop and clk(flop2flop)-defualt is not coming.
[image]

**but inside docker in sta** 
i still not getting any other pathgroup except input2output pathgroup
[image]

closed 2023-07-15 by Vinayakamk