The-OpenROAD-Project/OpenSTA #173: App crashes when different types of start points specified for "report_checks -from"

Blankgithub, 2023-06-28:
Reproduce steps:
1. Modify last line of "examples/delay_calc.tcl" as "report_checks -from {clk u2}"
2. run OpenSTA with command "../app/sta delay_calc.tcl"
3. The app crashes.

ps: it seems that Sdc::makeFilterPath() could return a deleted exception pointer （in addException(exception)）

jjcherry56, 2023-06-28:
The issue is the the from mixes different object types (clock and instance) which the implementation isn't designed to handle. -from u2 won't work either because it is not
a start point (it needs to use -through).

jjcherry56, 2023-06-29:
fixed in commit 
8ed0b25 report_checks -from {clk inst} 
in https://github.com/parallaxsw/OpenSTA.git

closed 2023-06-29 by jjcherry56

Blankgithub, 2023-06-29:
> The issue is the the from mixes different object types (clock and instance) which the implementation isn't designed to handle. -from u2 won't work either because it is not a start point (it needs to use -through).

Thank you for response, James.

Do we have any plan to support mixed different object types in the future?