The-OpenROAD-Project/OpenSTA #175: STA not recognising the paths

Vinayakamk, 2023-06-30:
in the generated synthesis .v file, i found some of the flipflops connected to ouptut pins,(flipflop to output)
but while pathgroup  flops to output ,sta tells no paths found
why? and how can i fix it.

vijayank88, 2023-06-30:
@Vinayakamk 
Attaching test case will be easy to debug.

Vinayakamk, 2023-06-30:
for the synthesised multiplier design (mk_combo_mul),i wanted to do path group.im struggling hard to do path groups (input to output ,flop to output,input to flop,and clk) 

i succesfully able to do all these path groups for other design but now for this,can u plz kindly check whats happening. 

[mk_combo_mul.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/11918764/mk_combo_mul.txt) {this is yosys generated synthesied verilog file}
in the file attched,from line number 48789,there are registers which are connected to output.

but find_timing_paths -from { list of all registers} -to {list of all inputs } says no paths found.

it would be great help if i could reach you online ,

vijayank88, 2023-07-01:
Have you tried `-through` option?

Vinayakamk, 2023-07-01:
no.because to use -through i scare if any paths are neglected.
since i able to do all pathgroups for picorv32 without mentioning -through ,i dint go for using it.
i will again try as you suggested.

closed 2023-07-15 by Vinayakamk