The-OpenROAD-Project/OpenSTA #176: Problem in reading in CCS Liberty files

gburdo, 2023-07-02:
Hi.
I tried to load TSMC 12nm Standard Cells libraries .lib files into sta and I get the following Warnings:
Warning: /tools/technology/TSMC/12nm/12FFC-LL/STD_CELLs/TSMCHOME/digital/Front_End/timing_power_noise/CCS/tcbn12ffcllbwp16p90_100a/tcbn12ffcllbwp16p90tt0p8v25c_ccs.lib line 631, table template ccsn_vout_template not found.
Warning: /tools/technology/TSMC/12nm/12FFC-LL/STD_CELLs/TSMCHOME/digital/Front_End/timing_power_noise/CCS/tcbn12ffcllbwp16p90_100a/tcbn12ffcllbwp16p90tt0p8v25c_ccs.lib line 649, table template ccsn_prop_template not found.
Warning: /tools/technology/TSMC/12nm/12FFC-LL/STD_CELLs/TSMCHOME/digital/Front_End/timing_power_noise/CCS/tcbn12ffcllbwp16p90_100a/tcbn12ffcllbwp16p90tt0p8v25c_ccs.lib line 1047, table template ccs_template not found.

I get a lot of these warnings.
What is interesting is that when I load a similar file (TSMC 12nm) but a smaller one I don't get these warnings.

I checked the .lib files and the tables templates are defined in it:
[image]
[image]
[image]

It also looks like sta doesn't get timing constraints information from the .lib file (probably warnings I get indicate it) because when I define a clock in my design I don't get any timing paths while I know that there are such (appear in PrimeTime).

Unfortunately I cannot send the TSMC 12nm .lib files for debug since I'm not allowed to do that.
Any clue to what to look at?

Thanks a lot
Gennady

jjcherry56, 2023-07-02:
what is the git commit of the version you are running?
Some issues with css noise libraries (which it looks like what you are reading) were fixed a few months ago.
ccs delay calculation is not supported in the public version of opensta but the libraries should read.
I don't have tsmc12 libraries but I the pdk may include nldm libraries you can use instead.

If you REALLY want to use those libraries you should be able to generate a testcas that includes the relevant groups in one of the public libraries in opensta/examples but no numbers from tsmc to reproduce the issue.

gburdo, 2023-07-03:
Hi jjcherry56.

Thanks a lot for your quick response.
I did follow this to install OpenSTA:
[image]

I will download the NDLM libraries and try using them. Will update you.
I will try to create a test-case with modified CCS libraries and then will post it as well.

Best Regards

gburdo, 2023-07-04:
Hi.
I tried using NDLM library files. With NDLM no warnings.
Got timing arcs. Didn't check yet if I see all expected arc.

I did try to write out my design .lib file with
I got 2 scenarios:
1. Crash with core dump
2. STA is getting stuck for hours and consumes more and more memory.

I have 64GB RAM so I would not expect any RAM issues with my small test-case design,

This is the script I tried to run:
[image]

Thanks a lot

jjcherry56, 2023-07-05:
"intalling with cmake" does not really answer the question of what commit you are running because the repo could be months old. 'git rev-parse HEAD' or git log tells you the commit
you have checked out.

write_timing_model may be blowing up because the design has many inputs combinationally 
connected to outputs which is by definition has an polynomial run time. But that is a separate
issue and one I could not possibly address without a testcase that has all of the files to run it.

closed 2024-08-12 by maliberty