The-OpenROAD-Project/OpenSTA #208: consider publishing new tagged releases

proppy, 2023-12-15:
I'm trying to get `opensta` packaged for conda-forge and they recommend building from source tarball, see https://conda-forge.org/docs/maintainer/adding_pkgs.html#tarballs-no-repos

I'm curious if you've considered publishing tags on this repository and a regular cadence?

I found one tagged release here https://github.com/The-OpenROAD-Project/OpenSTA/releases/tag/v2.2.0 but it looks like there were many changes since then https://github.com/The-OpenROAD-Project/OpenSTA/commits/master/

jjcherry56, 2023-12-15:
I gave up on bothering with tags a long time ago.
They might be useful when maintaining multiple release branches but I haven't had any reason
to have to deal with that overhead in a long time.
The version (in the cmake file) changes when  there are significant command or api changes.
The git hash is a lot more accurate "version" to indicate what sources someone is using.

proppy, 2023-12-19:
@hzeller has some automation in place at https://github.com/chipsalliance/verible/releases to tag release periodically, could this be an acceptable solution here?

commit b8b67ba4 (2025-02-12): spef unescaped names resolves #208

commit ce315777630daa66702dc08f60992dd50ad822c1
Author: James Cherry <cherry@parallaxsw.com>
Date:   Mon Feb 10 17:30:34 2025 -0700

    spef missing escapes

    Signed-off-by: James Cherry <cherry@parallaxsw.com>
closed 2024-08-12 by maliberty