The-OpenROAD-Project/OpenSTA #217: Bug: Or (||) filter expressions not working properly

akashlevy, 2024-01-24:
I'm looking for reset ports in `picorv32` design synthesized on `sky130hd`. There is one reset port called `resetn`. However, I get different results for the following two commands:
```
% get_ports -filter ((name=~rst*)||(name=~reset*)) *
% get_ports -filter ((name=~reset*)||(name=~rst*)) *
_6042a024dd7f0000_p_Port
```
These commands should obviously give the same results.

Using latest version of OpenSTA 2.4.0. Happy to provide test files to reproduce if needed.

jjcherry56, 2024-04-14:
If you had spaces around the || as the documentation indicates it would have worked.
I fixed it to not require the spaces (which means the filter has to be in quotes) in 
https://github.com/parallaxsw/OpenSTA.git commit
4146e623 get_* -filter do not require spaces around ||/&&

But since the whole point of get_ports is to pattern match the port names there isn't much point in using -filter arg.

[concat [get_ports rst*] [get_ports reset*]]

akashlevy, 2024-04-14:
Thanks James, this is helpful

closed 2024-04-14 by akashlevy