The-OpenROAD-Project/OpenSTA #235: Question regarding $width annotation

mole99, 2024-03-29:
Hello!

I noticed that SDF files written with OpenROAD and OpenSTA using the Sky130 PDK do not contain width annotations.

Here is an example from an SDF file:

```
 (CELL
  (CELLTYPE "sky130_fd_sc_hd__dfrtp_1")
  (INSTANCE _000_)
  (DELAY
   (ABSOLUTE
    (IOPATH CLK Q (0.324:0.324:0.324) (0.360:0.360:0.360))
    (IOPATH RESET_B Q () (0.000:0.000:0.000))
   )
  )
  (TIMINGCHECK
    (REMOVAL (posedge RESET_B) (posedge CLK) (0.336:0.336:0.336))
    (RECOVERY (posedge RESET_B) (posedge CLK) (-0.211:-0.211:-0.211))
    (HOLD (posedge D) (posedge CLK) (-0.045:-0.045:-0.045))
    (HOLD (negedge D) (posedge CLK) (-0.048:-0.048:-0.048))
    (SETUP (posedge D) (posedge CLK) (0.071:0.071:0.071))
    (SETUP (negedge D) (posedge CLK) (0.117:0.117:0.117))
  )
 )
```

The Verilog model for the cell `sky130_fd_sc_hd__dfrtp_1` has support for $width timing checks:

```
`celldefine
module sky130_fd_sc_hd__dfrtp_1 (
    Q      ,
    CLK    ,
    D      ,
    RESET_B,
    VPWR   ,
    VGND   ,
    VPB    ,
    VNB
);

    // Module ports
    output Q      ;
    input  CLK    ;
    input  D      ;
    input  RESET_B;
    input  VPWR   ;
    input  VGND   ;
    input  VPB    ;
    input  VNB    ;

    // Local signals
    wire buf_Q          ;
    wire RESET          ;
    reg  notifier       ;
    wire D_delayed      ;
    wire RESET_B_delayed;
    wire CLK_delayed    ;
    wire awake          ;
    wire cond0          ;
    wire cond1          ;

    //                                  Name  Output  Other arguments
    not                                 not0 (RESET , RESET_B_delayed                                    );
    sky130_fd_sc_hd__udp_dff$PR_pp$PG$N dff0 (buf_Q , D_delayed, CLK_delayed, RESET, notifier, VPWR, VGND);
    assign awake = ( VPWR === 1'b1 );
    assign cond0 = ( awake && ( RESET_B_delayed === 1'b1 ) );
    assign cond1 = ( awake && ( RESET_B === 1'b1 ) );
    buf                                 buf0 (Q     , buf_Q                                              );

specify
( negedge RESET_B => ( Q +: RESET_B ) ) = ( 0:0:0 ) ;  // delay is tris
( posedge CLK => ( Q : CLK ) ) = ( 0:0:0 , 0:0:0 ) ; // delays are tris , tfall
$recrem ( posedge RESET_B , posedge CLK , 0:0:0 , 0:0:0 , notifier , awake , awake , RESET_B_delayed , CLK_delayed ) ;
$setuphold ( posedge CLK , posedge D , 0:0:0 , 0:0:0 , notifier , cond0 , cond0 , CLK_delayed , D_delayed ) ;
$setuphold ( posedge CLK , negedge D , 0:0:0 , 0:0:0 , notifier , cond0 , cond0 , CLK_delayed , D_delayed ) ;
$width ( posedge CLK &&& cond1 , 1.0:1.0:1.0 , 0 , notifier ) ;
$width ( negedge CLK &&& cond1 , 1.0:1.0:1.0 , 0 , notifier ) ;
$width ( negedge RESET_B &&& awake , 1.0:1.0:1.0 , 0 , notifier ) ;
endspecify
endmodule
`endcelldefine
```

Since `SdfWriter::writeTimingChecks` in OpenSTA seems to have support for width annotations, I suspect that the problem lies in the data provided by the PDK.

I would like to ask if you perhaps know what data the PDK needs to provide in order for the width checks to get annotated?

Thank you!

jjcherry56, 2024-04-14:
There are 2 ways to specify min_pulse_width in a liberty file; with the port min_pulse_width_high/low attribute or with a timing arc of type min_pulse_width. The sky130 liberty files uses the latter form, which OpenSTA does not currently support. It should.

mole99, 2024-04-15:
Thanks for the info!

RTimothyEdwards, 2024-04-16:
@jjcherry56 :  Any chance that's on the roadmap for OpenSTA development?  If not, is it feasible to convert the sky130 PDK liberty files from using the min_pulse_width timing arc to using the min_pulse_width_high and min_pulse_width_low attributes?

jjcherry56, 2024-04-17:
The https://github.com/parallaxsw/OpenSTA.git repo commit
60d8030a liberty min_pulse_width timing group support
adds support for this flavor of min_pulse_width

mole99, 2024-04-18:
Awesome, thank you!

commit b87ec3fe (2025-04-14): linux compile resolves #235
closed 2024-08-12 by maliberty