The-OpenROAD-Project/OpenSTA #237: Write verilog doesn't reflect interconnected power ports

smunaut, 2024-04-22:
You can have multiple power / ground ports connected to each other in the database and this will not be reflected in the written verilog.
This results in LVS errors later in the flow because the powered netlist written by openroad doesn't math the extracted spice netlist.

I applied this path to fix it locally :

```difff
diff --git a/verilog/VerilogWriter.cc b/verilog/VerilogWriter.cc
index e3362da..51cf317 100644
--- a/verilog/VerilogWriter.cc
+++ b/verilog/VerilogWriter.cc
@@ -214,9 +214,9 @@ VerilogWriter::verilogPortDir(PortDirection *dir)
   else if (dir == PortDirection::bidirect())
     return "inout";
   else if (dir == PortDirection::power())
-    return "input";
+    return "inout";
   else if (dir == PortDirection::ground())
-    return "input";
+    return "inout";
   else if (dir == PortDirection::internal())
     return nullptr;
   else {
@@ -412,7 +412,7 @@ VerilogWriter::writeAssigns(Instance *inst)
     Net *net = network_->net(term);
     Port *port = network_->port(pin);
     if (port
-        && network_->direction(port)->isAnyOutput()
+        && (network_->direction(port)->isAnyOutput() || network_->direction(port)->isPowerGround())
         && !stringEqual(network_->name(port), network_->name(net))) {
       // Port name is different from net name.
       string port_vname = netVerilogName(network_->name(port),
```

maliberty, 2024-04-22:
Its best to file this against the upstream https://github.com/parallaxsw/OpenSTA

closed 2024-04-22 by maliberty

smunaut, 2024-04-23:
My bad, I thought this _was_ upstream ... I wasn't aware it was a clone/fork since it's not shown as such on github.

maliberty, 2024-04-23:
Sorry, it isn't obvious.

smunaut, 2024-04-24:
It's been fixed upstream FWIW