The-OpenROAD-Project/OpenSTA #239: Limit data-to-data pin skew command

phsauter, 2024-05-03:
Does OpenSTA have a command similar to Vivados `set_bus_skew`?

This command can limit the difference in arrival time between two disconnected data pins. 
This is useful to ensure a group of signals going through a clock domain crossing arrive within a certain time of each other.

https://www.intel.com/content/www/us/en/docs/programmable/683562/21-3/clock-domain-crossing.html

rovinski, 2024-05-04:
`set_bus_skew` is not an SDC command and isn't supported. You won't find that command in any ASIC tools. Here is some recommended reading on clock domain crossing constraints in ASICs: https://gist.github.com/brabect1/7695ead3d79be47576890bbcd61fe426

phsauter, 2024-05-04:
Thats a fantastic write-up thank you very much, our current constraints are a mix of what over people did before me, I will unify them into the described method.  

I know that `set_bus_skew` is not an SDC command and just wanted to see if there is something similar.
SDC has `set_data_check` which may be used for the same purpose but yesterday I read the OpenSTA documentation and wasn't quite convinced it is implemented the same way. I think I was just tired, reading it to day its pretty clear that it implements the same check.  
Still, I am glad I asked, otherwise I wouldn't have known about the CDC writeup you linked, thanks for that.

I am closing this issue.

closed 2024-05-04 by phsauter