The-OpenROAD-Project/OpenSTA #242: The command report_dcalc returns no results

RENEK-bool, 2024-06-02:
After sourcing example1.tcl, I want to see how the timing arc between r1/Q to u2/A1 is calculated, so I enter the command 
```
report_dcalc -from r1/Q -to u2/A1
```
but nothing is returned

akashlevy, 2024-08-12:
Can you attach a test case? It's impossible to debug based on what you have written here...

RENEK-bool, 2024-08-12:
I downloaded OpenSTA-2.2.0 from here https://github.com/The-OpenROAD-Project/OpenSTA/releases/tag/v2.2.0
There is an examples folder inside as shown below.
[image]
The content of example1.tcl is as follows
```
# sdf example
read_liberty example1_slow.lib
read_verilog example1.v
link_design top
read_sdf example1.sdf
create_clock -name clk -period 10 {clk1 clk2 clk3}
set_input_delay -clock clk 0 {in1 in2}
report_checks
```
As mentioned in my question, after I sourced example1.tcl in OpenSTA, I further entered the command report_dcalc -from r1/Q -to u2/A1, and nothing was returned.
The following is the content of example1.v
```
module top (in1, in2, clk1, clk2, clk3, out);
  input in1, in2, clk1, clk2, clk3;
  output out;
  wire r1q, r2q, u1z, u2z;

  DFF_X1 r1 (.D(in1), .CK(clk1), .Q(r1q));
  DFF_X1 r2 (.D(in2), .CK(clk2), .Q(r2q));
  BUF_X1 u1 (.A(r2q), .Z(u1z));
  AND2_X1 u2 (.A1(r1q), .A2(u1z), .ZN(u2z));
  DFF_X1 r3 (.D(u2z), .CK(clk3), .Q(out));
endmodule // top
```

RENEK-bool, 2024-08-12:
I also tried source example2.tcl. The difference between example1.tcl and example2.tcl is that example2.tcl does not have this command read_sdf example1.sdf. But it seems to have no effect. The command report_dcalc does not return any information.
[image]

This is the content of example2.tcl
```
# delay calc example
read_liberty example1_slow.lib
read_verilog example1.v
link_design top
create_clock -name clk -period 10 {clk1 clk2 clk3}
set_input_delay -clock clk 0 {in1 in2}
report_checks

```

closed 2024-08-12 by maliberty