The-OpenROAD-Project/OpenSTA #274: Rename `BUILD` to `BUILD.bazel`?

donn, 2025-06-26:
This isn't upstream - `BUILD` collides with the de-facto default folder for cmake builds, `build`, on case-insensitive filesystems.

closed 2025-07-31 by maliberty

donn, 2025-08-01:
@maliberty i think that resolves the upstream #274, not this one

maliberty, 2025-08-01:
There is no upstream BUILD file afaik.  I don't get the comment.

donn, 2025-08-01:
Let me clarify:

- This fork has a file named `BUILD` that does not exist upstream. This conflicts with `build` for cmake.
- You recently synchronized https://github.com/The-OpenROAD-Project/OpenSTA/commit/959f2dbb6ad3625c79d303471cc122deda5a5352, which contains the string "resolves #274"
- It locked *this* issue even though it referred to ParallaxSW/OpenSTA#274

maliberty, 2025-08-01:
Are you trying to build this repo as source of standalone sta?  It is generally just used as a submodule for OR where this doesn't matter.  It mattered in OR because that is the top of the build tree.